revilor/Marlin #1: M600 - 'PRUSA_MMU2_FILAMENT_RUNOUT_SCRIPT' was not declared in this scope

dcwalmsley, 2019-01-13:
Good day,
I downloaded you latest code and attempted to compile it using my parameter settings within the Configuration.h and Configuration_Adv.h file.  This was the message I received.
----------------------------------------
In file included from C:\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:28:0,
  from c:\users\dcwal\appdata\local\temp\arduino_build_82233\sketch\src\hal\hal_avr\hal.h:27,
  from c:\users\dcwal\appdata\local\temp\arduino_build_82233\sketch\src\inc/MarlinConfig.h:30,
  from c:\users\dcwal\appdata\local\temp\arduino_build_82233\sketch\src\marlin.h:24,
 from C:\Users\dcwal\AppData\Local\Temp\arduino_build_82233\sketch\src\feature\prusa_MMU2\mmu2.cpp:23:
C:\Users\dcwal\AppData\Local\Temp\arduino_build_82233\sketch\src\feature\prusa_MMU2\mmu2.cpp: In member function 'void MMU2::filamentRunout()':
C:\Users\dcwal\AppData\Local\Temp\arduino_build_82233\sketch\src\feature\prusa_MMU2\mmu2.cpp:730:36: error: 'PRUSA_MMU2_FILAMENT_RUNOUT_SCRIPT' was not declared in this scope
   enqueue_and_echo_commands_P(PSTR(PRUSA_MMU2_FILAMENT_RUNOUT_SCRIPT));
--------------------------------------

Going back through all the files, I noticed there isn't the statement located anywhere.  Should I modify line "#define FILAMENT_RUNOUT_SENSOR" in the Configuration.h file to "PRUSA_MMU2_FILAMENT_RUNOUT_SCRIPT" and see if this compiles?

~Doug Walmsley

revilor, 2019-01-13:
Looks like you got the commit where was still the wrong define in Configuration_adv.h. Sorry for that. Just add

#define PRUSA_MMU2_FILAMENT_RUNOUT_SCRIPT "M600"
(you will also need ADVANCED_PAUSE_FEATURE then)

or 

#define PRUSA_MMU2_FILAMENT_RUNOUT_SCRIPT ""
(filament runout handling will be disabled)

and it should compile.

dcwalmsley, 2019-01-14:
I download your latest files and issue has been resolved.

closed 2019-01-14 by dcwalmsley