revilor/Marlin #2: Request validation of my Configuration

dcwalmsley, 2019-01-14:
Background

HyperCube Core XY printer heated bed and Prusa Mk3 Direct Drive Extruder with optical filament detection sensor and the new MMU2 unit installed.

MMU2 can succesfully prime all 5 input spools to MMU sensor and back to staging point.

I have successfully flashed today's build with what I believe the correct configurations to support the MMU2.

Problem is that nothing I do via Ponterface or from LCD yields any indication that I'm talking to the MMU2 unit.  I tried loading one then all filaments from the LCD menu and nothing happened.  I noticed that no activity from the MMU is occurring and I'm suspecting that I haven't configured something correctly in reference to the MUX pins.

In Configuration.h there is a section for Prusa's MMU which I do not have enabled as this was code put in place back when MMU1 was the only multiplexer out from Prusa.

In the Configuration.h file I selected #define MOTHERBOARD BOARD_EINSY_RAMBO for my motherboard.  In the Pins Folder, I noticed in "pins_EINSY_RAMBO.h" that the 
// Průša i3 MK2 Multiplexer Support
//
#define E_MUX0_PIN         17
#define E_MUX1_PIN         16
#define E_MUX2_PIN         78
//
According to Einsy "Schematic Prints_Einsy Rambo_1.1a" pin assignments for EXP3 (or P3 on schematics) where the MMU 2 is plugged to, the E_MUX2 PIN is connected to Reset (9) of the PJ5 (76).

--Change--
// Průša i3 MK2 Multiplexer Support
//
#define E_MUX0_PIN         17
#define E_MUX1_PIN         16
#define E_MUX2_PIN         76   //Changed from 78 to 76 as per PJ5 pin connection.
//

So my question is, do I need to uncomment the three MUX pins in the Configuration.h file on line #'s 172-174?

At this point Neither MMU2 nor the Prusa filament sensor functioning at this time.

I've zipped and attached the following files for your review.
Configuration.h
Configuration_Adv.h
pins_EINSY_RAMBO.h
and
Schematic Prints_Einsy Rambo_1.1a.PDF

I'd be grateful if you could review what I configured and point me in the right direction.

Best Regards,
Doug Walmsley

[DCW MMU2 HyperCube Files.zip](https://github.com/revilor/Marlin/files/2753905/DCW.MMU2.HyperCube.Files.zip)

dcwalmsley, 2019-01-14:
Have some success in communicating with MMU2 via LCD control panel.  Seems that I was able to get it to move to the 4th filament port and pull filament to FINDA sensor then retract to staging point.  I then asked it to load to nozzle and it extended 425mm of filament down the PTFE tubing.  

Although the MMU2 unit is moody and I'm not able to get same results, this is the first time I can say that I was able to communicate with it.

revilor, 2019-01-15:
The E_MUX*_PINs are for the Prusa MMU V1 (the one with 4 extruder steppers and the Y-splitter on top of the extruder). The MMU2 does not use those so it does not matter what values the defines have.

From the schematics of the Einsy board I would say you have to 

#define PRUSA_MMU2_RST_PIN 76

Without that Marlin will not be able to reset the MMU2 on startup.

If you have no other filament sensor than the MMUs FINDA , remove FILAMENT_RUNOUT_SENSOR

Remains the issue "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN." which is not related to the MMU2.

dcwalmsley, 2019-01-16:
@revilor,

Thanks again for reviewing my configuration.  So aside from the FINDA sensor on the MMU, I also have the Prusa optical sensor above the direct drive.  Should I keep "FILAMENT_RUNOUT_SENSOR" if I'm going to use this sensor?  If not, I understand that this is another item you are looking to get working within Marlin.

revilor, 2019-01-17:
We don't have support for the Prusa optical sensor in Marlin at the moment. So I would disable FILAMENT_RUNOUT_SENSOR.

dcwalmsley, 2019-01-17:
Thanks again for reviewing my configurations.  I've disabled all MUX related lines and ensured pin 76 was called for "#define PRUSA_MMU2_RST_PIN".  

I've disabled the FILAMENT_RUNOUT_SENSOR for now.

closed 2019-01-17 by dcwalmsley