revilor/Marlin #3: MMU2 Filament Length too Short

dcwalmsley, 2019-01-14:
MMU2 controlled from LCD panel.   I load filament and was successfull. I then asked it to load to nozzle and it extended 425mm of filament down the PTFE tubing.  

My current issue is how to get it to extend 570mm of PTFE tubing as I'm not using the supplied PRUSA orange (360mm) tubing and I need to be able to feed all the way to the direct drive.

Where should I look to make adjustment to feeding length?

~Doug

revilor, 2019-01-15:
This has to be done in the MMU, see [https://prusa3d.com/downloads/manual/prusa3d_manual_mmu2_eng_1_03.pdf](https://prusa3d.com/downloads/manual/prusa3d_manual_mmu2_eng_1_03.pdf) page 15ff

dcwalmsley, 2019-01-15:
Thanks. I will attempt this tomorrow after work.

dcwalmsley, 2019-01-16:
@revilor,

After looking through your comments and doing a bit more search, I came across one of the authors (Ruud Rademaker - https://github.com/ruedli/MM-control-01/tree/load_reload_using_bowdenlength ) for the update to v1.0.3.  He submitted and now is part of the latest version of the MMU firmware the ability to calibrate up to 792mm which is more than enough for me to finish my calibration.  I also pinged him to see if his faster feed and retraction speeds are also part of Pursa's release.  With a longer PTFE tubing, speeding up the loading and unloading will cut down on print time when performing multiple material printing.

Thanks again for pointing me in the right direction.