revilor/Marlin #5: Issue: MMU Resetting while Extruder is Cold and Filament is in Direct-Drive

dcwalmsley, 2019-01-17:
My MMU (FW 1.0.3) is behaving on reset although I'm trying to figure out what to do when a reset has occured while extruder is cold and direct-drive is loaded.  

I noticed the MMU attempts to pull back any filament that has been triggered via the FINDA sensor before the MMU performs a drive selector calibration and repositioning.  If the filament is loaded all the way to extruder and direct-drive, the filament feed drive motor in MMU can be heard grinding on the filament in its attempt to withdraw the filament past the FINDA sensor.

So here I am asking the question...  How do I get the MMU to withdraw filament under this situation?  
One thought is to ensure the filament is retracted manually before a reset occurs.  This is not the practical solution as a reset can occur from cut power, pressing reset button, software connections, etc.

Another thought is to incorporate a process via LCD, G-Code, or Firmware which allows for a sanity check of filament already loaded under cold situations.  
1. During Power-on, reset or software connection, the firmware senses temperature of nozzle and FINDA or filament sensor and;
   a. Provides alert message that MMU is unable to retract until temperature has been set for retraction.
   b. LCD menu alert operator with a "Do you want nozzle to heat up?  If so, last temperature used? If you choose yes to this option the control board then heats to last printed temperature.  Once this has been achieve, then direct-drive pulls filament back to release point before MMU filament feed drive motor engages and retracts the filament.  If you choose no, then the nozzle temperature window opens and you must dial the desired temperature with the same end result.  Maybe a third option should be added too.  This one allows operator to cancel this sanity causing the MMU to attempt to retract regardless of where the filament is.

Under current situations if  MMU reinitialize and attempts to pull back filament could cause issues with damaging filament in MMU, leaving particles that could lead to jamming, and failed prints.

I'm open to ideas and suggestions on improving the unloading/loading process of MMU during reset but this is an annoyance as this moment and time.

Thanks.

~Doug W

revilor, 2019-01-18:
I would say if the MMU moves the pulleys on startup or reset then there is something wrong with your MMU.

From what I have seen so far the MMU with firmware 1.0.3 only moves/homes the idler drum after reset or power on. The selector homes after the MMU got the first tool change command from the printer or the MMU buttons, the pulleys move only when filament has to be transported. And in case there is filament detected by the FINDA when homing the selector, the MMU does not try to unload the filament but just switches to alarm mode with all red LEDs blinking. 

So actually the scenario you describe is already taken care of in the MMU firmware since 1.0.0.

dcwalmsley, 2019-01-30:
So I decided to perform at end of print a G-Code to withdraw the filament from the extruder prior to shutting down the heater.  This seems to fix part of the issue as I still have to worry if the printer has a kill function and cools down that the filament cannot be retracted by default when printer has been re-initialized or a connection re-established.  

My thought is that there needs to be a check in the firmware that identifies that the filament has not been retracted past direct-driver extruder and initiates a heater warm up then performs a MMU retraction.  Since the machine has no idea what temp is needed, there needs to be a menu alert to set appropriate temperature before MMU executes retraction.

closed 2019-02-03 by dcwalmsley